NethServer/arm-dev #10: Nextcloud is not in Development

markVnl, 2018-07-21:
Due to php7 requirement and unfamiliarity with Nextcloud it is not in dvelopment. 

There are some serious people working on php 7.2 for centos-arm32 and therefor do not think it’s impossible. 

If there is demand and someone adopts it for testing we can try to port it to arm32 bit.

markVnl, 2018-07-26:
functional prove of concept running it on php72-fpm  

See: https://community.nethserver.org/t/php72-scl-on-raspberrypi-for-nextcloud/10403

After cleaning up of ns-scl-testing repo it can be moved to "in development"

markVnl, 2018-07-29:
https://github.com/markVnl/nethserver-arm-dev/commit/627f985d2e73b1915b7540d4f125861c9a532e87

https://github.com/markVnl/nethserver-arm-php72-php-fpm/commit/38917f51f29db4be8f090728ae54e16ab8a65865

https://github.com/markVnl/nethserver-nextcloud/commit/0e58086145dfeaaac58a60384972d558ab247e38

DavidePrincipi, 2019-05-16:
As there are no updates I close this issue. Please reopen it if necessary.

closed 2019-05-16 by DavidePrincipi