NethServer/arm-dev #16: Web proxy

markVnl, 2018-07-21:
Module still needs a good shake and testing:

* Does it install without errors. 
* Does it function as expected.
* How does it behave.

mrmarkuz, 2018-07-24:
When you set auth proxy you get a red banner and following error in /var/log/messages:

FATAL: Authentication helper program /usr/lib64/squid/basic_pam_auth: (2) No such file or directory

Again it's a path problem with `/usr/lib64` which should be `/usr/lib` like in [OpenVPN](https://github.com/markVnl/nethserver-arm-dev/issues/19):

/etc/e-smith/templates/etc/squid/squid.conf/20acl_10_auth:

line 13:
`           $OUT .= "auth_param negotiate program /usr/lib64/squid/negotiate_kerberos_auth -i\n";`
line 19:
`       $OUT .= "auth_param basic program  /usr/lib64/squid/basic_pam_auth\n";`

**Proposal:**

Can we add a link from /usr/lib64 to /usr/lib so we don't need change any package that uses it?

markVnl, 2018-07-25:
> Proposal:
> 
> Can we add a link from /usr/lib64 to /usr/lib so we don't need change any package that uses it?

I think this is the easiest way, and its a good idea to put this in as a workaround until we think of an more elegant solution.

Thanx

markVnl, 2018-12-19:
We keep track of port progress on a wiki page

https://wiki.nethserver.org/doku.php?id=quality:qa_testing_new_version_of_ns7.x_for_armhfp

closed 2018-12-19 by markVnl