NethServer/arm-dev #19: OpenVPN

markVnl, 2018-07-21:
Module still needs a good shake and testing:

* Does it install without errors. 
* Does it function as expected.
* How does it behave.

mrmarkuz, 2018-07-24:
openvpn@host-to-net does not start when you set to use Username, password and certificate as authentication on openvpn roadwarrior server page. You get a red banner.

In /var/log/messages:
`PLUGIN_INIT: could not load plugin shared object /usr/lib64/openvpn/plugins/openvpn-plugin-auth-pam.so: /usr/lib64/openvpn/plugins/openvpn-plugin-auth-pam.so: cannot open shared object file: No such file or directory: No such file or directory (errno=2)`

I found that a wrong path is used in a template fragment. If you change line 11 in `/etc/e-smith/templates/etc/openvpn/host-to-net.conf/50security` to
`    $OUT.="plugin /usr/lib/openvpn/plugins/openvpn-plugin-auth-pam.so /etc/pam.d/login\n";`
it works without banner.

Tested routed roadwarrior with Username, password and certificate from Android device and it works as expected.

markVnl, 2018-07-27:
Can confirm: 
Tested routed roadwarrior with Username, password and certificate from windows 10 and it works as expected.

closed 2018-07-27 by markVnl