NethServer/arm-dev #28: Failure of templates expansion containing /usr/lib64 (part 2)

markVnl, 2018-12-21:
Some templates have configuration lines referring to /usr/lib64, obviously absent on arm32 bit .

known to be affected:

- nethserver-openvpn
- nethserver-webproxy

This is mitigated by adding a link from /usr/lib64 to /usr/lib so we did need change any package that uses it (See arm-dev https://github.com/markVnl/nethserver-arm-dev/issues/26, https://github.com/markVnl/nethserver-arm-dev/issues/16 and https://github.com/markVnl/nethserver-arm-dev/issues/19)

With the complete integration of arm in the nethsever-ecosystem it is preferred fo find a more elegant solution :)

DoTo open issues for 
-[ ] nethserver-openvpn
-[ ] nethserver-webproxy

DavidePrincipi, 2019-02-27:
@markVnl both related issues were closed: can we close this one too?

gsanchietti, 2019-03-06:
Yes, closing.

closed 2019-03-06 by gsanchietti