NethServer/arm-dev #29: restic: add arm support

dz00te, 2019-03-04:
while probably not so elegant,  can we add to spec of restic something like this to make multi arch ?

```
%ifarch %{arm}
%define nsarch arm
%endif
%ifarch aarch64
%define nsarch arm64
%endif
%ifarch x86_64
%define nsarch amd64
%endif
Source0: https://github.com/restic/restic/releases/download/v%{version}/restic_%{version}_linux_%{nsarch}.bz2
```

or i could upload in testing the build for aarch64 and armhfp?

gsanchietti, 2019-03-06:
I like the patch, it should be good for all environment.

Just open the PR, then you need to manually build/upload for aarch64 and armhfp (you should already have the permissions to do it).

dz00te, 2019-03-07:
ok, just do not get away, I might need some help :)

dz00te, 2019-03-07:
builded and uploaded packakages for armhfp and aarch64 in arm-testing

markVnl, 2019-08-30:
> builded and uploaded packakages for armhfp and aarch64 in arm-testing

Can we move the packages to updates?

gsanchietti, 2019-08-30:
Done ;)

closed 2019-08-30 by gsanchietti