NethServer/arm-dev #30: Missing sclo(-rh) repositories for arm breaks system

markVnl, 2019-08-30:
Subject kind of say it all..

Introduced around here  NethServer/dev#5704 ,
or more specific with enabling these repo's  if you have no subscription 

Lines of code to review:

https://github.com/NethServer/nethserver-release/blob/4c116c8dedd98d1a153063e956edebd03401d416/root/etc/e-smith/templates/etc/yum.repos.d/NethServer.repo/10base#L2

https://github.com/NethServer/nethserver-release/blob/4c116c8dedd98d1a153063e956edebd03401d416/root/etc/e-smith/templates/etc/yum.repos.d/NethServer.repo/10base#L62

Also see:

https://community.nethserver.org/t/arm-development-next-steps/11047/76
https://community.nethserver.org/t/installing-updates-on-arm-bananapi-fails-with-mirror-not-found/12206
https://community.nethserver.org/t/ultimate-ns7-software-updates-origin-policy/11691

gsanchietti, 2019-08-30:
We could try to create empty repositories to just not break the system.
Some packages will be still not available, but at least yum will work.

What do you think?

AFAIK ce-sclo-rh for aarch64 is good. The broken ones should be:

- ce-sclo-rh for armhfp
- ce-sclo-sclo for armhfp
- ce-sclo-sclo for aarch64

Can you confirm?

markVnl, 2019-08-30:
> What do you think?

funny you propose this, it croosed my mind and thought you would not agee on such a mitigation.. 

Already thought up some compelling  arguments as "it would keep the code clean an uniform accros all arch's" ...

So Yes

>     * ce-sclo-rh for armhfp
> 
>     * ce-sclo-sclo for armhfp
> 
>     * ce-sclo-sclo for aarch64
> 
> 
> Can you confirm?

Yes

gsanchietti, 2019-08-30:
Of course it's not elegant, but the modification should be limited to the infra, not a big deal.
I ask you just to wait a bit until Davide is back from holiday :)

DavidePrincipi, 2019-09-16:
I agree the proposed solution is appealing: return an empty repo from mirrorlist.

Let me advance another approach: disable the offending repositories in `/etc/nethserver/eorepo.conf`. It would require to add a static file in the ARM image under `/etc/e-smith/templates/etc/nethserver/eorepo.conf/`, which removes SCLo identifiers from the list.

markVnl, 2019-09-16:
> can't work: it requires also a custom template of `nethserver.repo`

One conditional in  `nethserver.repo` for arm is not my big concern, just exit (return) before scl repo's are expanded if arch is not x86_64.. - or something like this - . Quite positive we can figure-out a stable solution. =:)

The problematic scenario is a manual install on a system running default centOS., ie run `nethserver-instal` after having manualy installed nethserver-release.  Then it start's of with a "non templated" `nethserver.repo` hence no conditional possible ; This still works on arm because 
[packages.nethserver.org/nethserver/nethserver-release-7.rpm](packages.nethserver.org/nethserver/nethserver-release-7.rpm)
is not updated (and please do not do this right now)

gsanchietti, 2019-09-17:
> and please do not do this right now

The link will be updated only after 7.7 release :)

DavidePrincipi, 2019-09-17:
The PR was merged. Please @markVnl check it out!

markVnl, 2019-09-17:
Can confirm the solution works.

On armhfp (raspberry PI) after updating a fresh install:

```
# yum makecache
Loaded plugins: changelog, fastestmirror, nethserver_events
Loading mirror speeds from cached hostfile
 * ce-base: mirrors.coreix.net
 * ce-extras: mirrors.coreix.net
 * ce-updates: mirrors.coreix.net
ce-base/7/armhfp/signature                                                       |  811 B  00:00:00
ce-base/7/armhfp/signature                                                       | 3.6 kB  00:00:00 
ce-extras/7/armhfp/signature                                                     |  811 B  00:00:00
ce-extras/7/armhfp/signature                                                     | 2.9 kB  00:00:00 
ce-sclo-rh                                                                       | 2.9 kB  00:00:00
ce-sclo-sclo                                                                     | 2.9 kB  00:00:00
ce-updates/7/armhfp/signature                                                    |  811 B  00:00:00
ce-updates/7/armhfp/signature                                                    | 2.9 kB  00:00:00 
epel                                                                             | 3.8 kB  00:00:00
nethserver-base/7/armhfp/signature                                               |  836 B  00:00:00
nethserver-base/7/armhfp/signature                                               | 2.9 kB  00:00:00 
nethserver-updates/7/armhfp/signature                                            |  836 B  00:00:00
nethserver-updates/7/armhfp/signature                                            | 4.1 kB  00:00:00 
Metadata Cache Created
```

Note we fooled yum @ `ce-sclo-rh` and `ce-sclo-sclo` successfully

closed 2019-09-17 by markVnl