NethServer/arm-dev #31: Unapble to install or update rpm's on armhfp with 7.7.1908

markVnl, 2019-10-07:
Centos 7.7.1908 armhfp repositories have an extra gpg key as can be seen in the Centos-Base.repo (truncated): 

```
# cat /etc/yum.repos.d/CentOS-Base.repo

# CentOS-Base.repo
...
...
[base]
name=CentOS-$releasever - Base
mirrorlist=http://mirrorlist.centos.org/?release=$releasever&arch=$basearch&repo=os&infra=$infra
#baseurl=http://mirror.centos.org/altarch/$releasever/os/$basearch/
gpgcheck=1
enabled=1
gpgkey=file:///etc/pki/rpm-gpg/RPM-GPG-KEY-CentOS-7
       file:///etc/pki/rpm-gpg/RPM-GPG-KEY-CentOS-SIG-AltArch-Arm32
```

Note the additional RPM-GPG-KEY-CentOS-SIG-AltArch-Arm32 which is  absent in the (expanded)  /etc/yum.repos.d/NethServer.repo

Possible solutions:
a) distribute arch depended nethserver-release packages
b) expand /etc/yum.repos.d/NethServer.repo with some arch conditionals still have to do some (dirty) stuff with rpm macro's if installed with nethserver-install on centos base.
a/b) only distribute arch depended nethserver-release packages for installing on centos base, updates of nethserver-releaseare are arch independent

DavidePrincipi, 2019-10-07:
I'd go with solution 2, by changing only the template code. If file `RPM-GPG-KEY-CentOS-SIG-AltArch-Arm32` exists, expand its reference under `gpgkey=`.

markVnl, 2019-10-07:
> I'd go with solution 2,

Oke i'll work on this first so we can get a RC image out;

And then we figure out a solution for the "nethserver-install on centos base" scenario, I think we can magage this in the nethserver-install script. It turns out the system does not break after instlling nethserver-release befor running nethserver-install. It seems just to ignore the ce-base and ce-update declarations as long the Centos-... repo's are enabled

markVnl, 2019-10-07:
more or less the same applies to aarch64 from `CentoOS-Base.repo `(truncated):

```
[base]
name=CentOS-$releasever - Base
mirrorlist=http://mirrorlist.centos.org/?release=$releasever&arch=$basearch&repo=os&infra=$infra
#baseurl=http://mirror.centos.org/altarch/$releasever/os/$basearch/
gpgcheck=1
gpgkey=file:///etc/pki/rpm-gpg/RPM-GPG-KEY-CentOS-7
       file:///etc/pki/rpm-gpg/RPM-GPG-KEY-CentOS-7-aarch64
```

markVnl, 2019-10-09:
How do we proceed, should I open an issue in /dev referencing this to start QA? 

It is the vital missing part to proceed with making / publishing armhfp 7.7.1908 images , aarch64 needs a little work but is also close..

DavidePrincipi, 2019-10-09:
I've already tested it on x86_64. If it's ok on the ARM side I can proceed with a new RPM release.

markVnl, 2019-10-09:
As you might had noticed on the picture @ https://community.nethserver.org/t/teaser-nethserver-cockpit-on-arm/13543  it already states "NethServer release 7.7.1908 **(final)**"

Is was already installed and working , tested aach64 too

markVnl, 2019-10-10:
> It is the vital missing part to proceed with making / publishing armhfp 7.7.1908 images

a gentle bump; when can we expect a new nethserver-release to be released?

cc/ @DavidePrincipi @gsanchietti

markVnl, 2019-10-10:
> 
> 
> Shall we wait for #32?
> 
> I expect nethserver-cockpit is working within the ARM image...

The chicken or the egg paradox is, 
we cannot ask for tests for #32 simply because without this you cannot install a single package..

ATM I patch every install i make manually to be able to test, somehow the autobuild packages did not make it to the testing repository's....

DavidePrincipi, 2019-10-11:
In nethserver-updates:

- http://mirror.nethserver.org/nethserver/7.7.1908/updates/x86_64/Packages/nethserver-release-7-16.ns7.noarch.rpm
- http://mirror.nethserver.org/nethserver/7.7.1908/updates/armhfp/Packages/nethserver-release-7-16.ns7.noarch.rpm
- http://mirror.nethserver.org/nethserver/7.7.1908/updates/aarch64/Packages/nethserver-release-7-16.ns7.noarch.rpm

closed 2019-10-11 by DavidePrincipi