NethServer/arm-dev #32: Nethserver-cockpit is not installable

markVnl, 2019-10-07:
nethserver-cockpit requires nethserver-subscription and the latter requires puppet-agent which is not availible for armhfp (and probably not for aarch64).

Moreover we do not have a subscription model for arm which is a community effort.

Also See:
https://community.nethserver.org/t/nethserver-on-raspberry-pi/2851/178
https://community.nethserver.org/t/release-candidate-iso-7-7-1908-rc1/13482/26

DavidePrincipi, 2019-10-07:
Proposal:

- define an arch-dependant subpackage for nethserver-subscription, (say nethserver-subscription-inventory). The ARM implementation is just an empty RPM.
- on the UI side, "disable" the support activation by checking a file existence from nethserver-subscription-inventory

markVnl, 2019-10-07:
to be honest this stands not very high on my prio list..

For me any solution will do;
lets ask @federicoballarini and @mrmarkuz  too

DavidePrincipi, 2019-10-07:
> to be honest ...

:joy: I'll take care of implementing it

DavidePrincipi, 2019-10-08:
@markVnl if the package-split is good I can complete the implementation on the UI side. Please let me know!

markVnl, 2019-10-08:
> @markVnl if the package-split is good I can complete the implementation on the UI side. Please let me know!

Will do so.
Have a appointment right now, first thing tomorrow morning .

DavidePrincipi, 2019-10-09:
https://community.nethserver.org/t/teaser-nethserver-cockpit-on-arm/13543

Ok, I proceed with the UI changes

DavidePrincipi, 2019-10-09:
...after a brief analysis I think there could be no need to change the UI now because Nethesis servers do not provide aarch64 and armhfp mirrors. 

    curl 'https://mirrorlist.nethserver.com/?repo=base&arch=x86_64&nsversion=7.7.1908&usetier=no'
    curl 'https://mirrorlist.nethserver.com/?repo=base&arch=armhfp&nsversion=7.7.1908&usetier=no'

I'll ask Alessio to write in the right places that Nethesis support is provided only for x86_64 ATM...

DavidePrincipi, 2019-10-09:
New packages in nethserver-testing 7.7

markVnl, 2019-10-09:
> New packages in nethserver-testing 7.7

Do I understand correctly latest PR#26 should result in:
 
[image]

After an attempt to register a non_x86_64 system?

DavidePrincipi, 2019-10-09:
Yes it is the expected output! However we could improve the messages to clarify it is not expected to work with non x86_64 arches /cc @alefattorini @gsanchietti

DavidePrincipi, 2019-10-09:
**Test cases**

Try both on x86_64 and ARM architectures

1. the nethserver-support package from testing must upgrade existing packages by requiring/adding the new arch-dependant `nethserver-support-inventory` RPM
2. the nethserver-support-inventory RPM ships the inventory scripts and pulls in facter only for x86_64. For ARM builds the file list must be almost empty (apart from docs)
3. Check out the NS7 installation procedure based on CentOS
4. The subscription page always fail at the validation step on non x86_64 arches. It must work as usual on x86_64.

DavidePrincipi, 2019-10-09:
News from @nethbot :

- nethserver-subscription-inventory-3.5.0-1.2.g69d7e72.ns7.x86_64.rpm to 7.7.1908/testing (x86_64)
- nethserver-subscription-3.5.0-1.2.g69d7e72.ns7.noarch.rpm to 7.7.1908/testing (x86_64)
- nethserver-subscription-3.5.0-1.2.g69d7e72.ns7.noarch.rpm to 7.7.1908/testing (armhfp)
- nethserver-subscription-3.5.0-1.2.g69d7e72.ns7.noarch.rpm to 7.7.1908/testing (aarch64)
- nethserver-subscription-ui-3.5.0-1.2.g69d7e72.ns7.noarch.rpm to 7.7.1908/testing (x86_64)
- nethserver-subscription-ui-3.5.0-1.2.g69d7e72.ns7.noarch.rpm to 7.7.1908/testing (armhfp)
- nethserver-subscription-ui-3.5.0-1.2.g69d7e72.ns7.noarch.rpm to 7.7.1908/testing (aarch64)

markVnl, 2019-10-09:
Will build nethserver-support-inventory for armhfp asap (as in now) and upload it to  /testing (armhfp)

gsanchietti, 2019-10-09:
> However we could improve the messages to clarify it is not expected to work with non x86_64 arches

I think that for now is good enough ;)

DavidePrincipi, 2019-10-09:
@markVnl, please note I added some `sed` commands that fix the version number and the changelog of `nethserver-subscription-inventory.spec`. Those commands run automatically during the Travis build. Please remember to run them manually for the ARM release.

markVnl, 2019-10-09:
> @markVnl, please note I added some `sed` commands that fix the version number and the changelog of `nethserver-subscription-inventory.spec`. Those commands run automatically during the Travis build. Please remember to run them manually for the ARM release.

Oops to late already uploaded to arm-testing... going to correct it

DavidePrincipi, 2019-10-09:
No problem for the testing build! Just keep it in mind for the release to update.

markVnl, 2019-10-09:
actually even if you do this with an manual test-build (as just done) , `make-sprm` undo(es) the sed commands in the actual build.. ;)

markVnl, 2019-10-09:
> @markVnl, please note I added some `sed` commands that fix the version number and the changelog of `nethserver-subscription-inventory.spec`. Those commands run automatically during the Travis build. [...]

For the future we may autotomize this (for reference) two commands which produce the (empty) aarch64 nethserver-subscription-inventory package on x86_64 buildhost: 

```
mock -D "dist .ns7" --resultdir=. -r nethserver-7-x86_64 nethserver-subscription-inventory-3.5.0-1.2.g69d7e72.ns7.src.rpm --arch=aarch64
rpmbuild -bb --target=aarch64 nethserver-subscription-inventory.spec

```

markVnl, 2019-10-10:
> 
> 
> **Test cases**
> 
> Try both on x86_64 and ARM architectures
> 
>     1. the nethserver-support package from testing must upgrade existing packages by requiring/adding the new arch-dependant `nethserver-support-inventory` RPM

**n/a** for armhfp and aarch64 (initial package)
> 
>     2. the nethserver-support-inventory RPM ships the inventory scripts and pulls in facter only for x86_64. For ARM builds the file list must be almost empty (apart from docs)

**OKE** for armhfp and aarch64

> 
>     3. Check out the NS7 installation procedure based on CentOS
> 

not possible without #31

>     4. The subscription page always fail at the validation step on non x86_64 arches. It must work as usual on x86_64.

**OKE** armhfp and aarch64

DavidePrincipi, 2019-10-15:
Released in nethserver-updates:

- nethserver-subscription-inventory-3.5.1-1.ns7.x86_64
- nethserver-subscription-ui-3.5.1-1.ns7.noarch
- nethserver-subscription-3.5.1-1.ns7.noarch

@markVnl do you want to close this issue after releasing the `nethserver-subscription-inventory` package for ARM?

markVnl, 2019-10-15:
> do you want to close this issue after releasing the `nethserver-subscription-inventory` package for ARM?

I will upload packages late afternoon -evening-  and close this afterwards

markVnl, 2019-10-15:
Released in nethserver-updates **7.7** armhfp & aarch64 :

- [nethserver-subscription-inventory-3.5.1-1.ns7.armv7hl.rpm](http://packages.nethserver.org/nethserver/7.7.1908/updates/armhfp/Packages/nethserver-subscription-inventory-3.5.1-1.ns7.armv7hl.rpm)
- [nethserver-subscription-inventory-3.5.1-1.ns7.aarch64.rpm]( http://packages.nethserver.org/nethserver/7.7.1908/updates/aarch64/Packages/nethserver-subscription-inventory-3.5.1-1.ns7.aarch64.rpm)

closed 2019-10-15 by markVnl