NethServer/arm-dev #33: gpg-keys import can fail at system-init

markVnl, 2019-10-11:
Related: #31

software center fails with a message ending with:
`repomd.xml signature could not be verified for nethserver-base.`

Current workaround, on the command line:
`yum makecache -y`

we fixed this is the past here:
https://github.com/NethServer/nethserver-base/blob/master/root/etc/e-smith/events/actions/nethserver-base-import-repokeys#L25

My best guess is we should add the extra  armhfp /aarch key's here too:
https://github.com/NethServer/nethserver-base/blob/master/root/etc/e-smith/events/actions/nethserver-base-import-repokeys#L39

markVnl, 2019-10-11:
extra info:

```
# gpg /etc/pki/rpm-gpg/RPM-GPG-KEY-CentOS-SIG-AltArch-Arm32
pub  2048R/62505FE6 2015-11-27 CentOS AltArch SIG - Arm32 (https://wiki.centos.org/SpecialInterestGroup/AltArch/Arm32) <security@centos.org>
```
```

# gpg /etc/pki/rpm-gpg/RPM-GPG-KEY-CentOS-7-aarch64
pub  2048R/305D49D6 2015-07-28 CentOS AltArch SIG - AArch64 (http://wiki.centos.org/SpecialInterestGroup/AltArch/AArch64) <security@centos.org>

```

markVnl, 2019-10-11:
Sorry I need to ping you @DavidePrincipi ;)

I do not understand where the hash/key (for instance) of the RPM-GPG-KEY-NethServer-7 relates to [here](https://github.com/NethServer/nethserver-base/blob/bb0c952c850018bc792457dbe6c30cc035bf7f0e/root/etc/e-smith/events/actions/nethserver-base-import-repokeys#L56)

DavidePrincipi, 2019-10-11:
It is the key fingerprint: you can obtain it with

    gpg --list-keys --fingerprint --with-colons </etc/pki/rpm-gpg/RPM-GPG-KEY-NethServer-7  | grep ^fpr

It needs to be hardcoded, so we are sure we do not trust  unknown keys. You could add the ARM-SIG keys there to avoid `yum -y` workaround...

DavidePrincipi, 2019-10-16:
**VERIFIED**

Updated an existing x86_64 installation: **OK**

Tested "Install on CentOS" installation on 7.7: **OK** (see log evidence of key import. Missing keys are ignored - `inserting ownertrust` lines)

```text
Oct 16 10:22:35 vm9 esmith::event[6765]: gpg: keyring `/tmp/tmp.HtHfQRbsU8/secring.gpg' created
Oct 16 10:22:35 vm9 esmith::event[6765]: gpg: keyring `/tmp/tmp.HtHfQRbsU8/pubring.gpg' created
Oct 16 10:22:35 vm9 esmith::event[6765]: gpg: /tmp/tmp.HtHfQRbsU8/trustdb.gpg: trustdb created
Oct 16 10:22:35 vm9 esmith::event[6765]: gpg: key F4A80EB5: public key "CentOS-7 Key (CentOS 7 Official Signing Key) <security@centos.org>" imported
Oct 16 10:22:35 vm9 esmith::event[6765]: gpg: key B6792C39: public key "CentOS-7 Debug (CentOS-7 Debuginfo RPMS) <security@centos.org>" imported
Oct 16 10:22:35 vm9 esmith::event[6765]: gpg: key F2EE9D55: public key "CentOS SoftwareCollections SIG (https://wiki.centos.org/SpecialInterestGroup/SCLo) <security@centos.org>" imported
Oct 16 10:22:35 vm9 esmith::event[6765]: gpg: key 8FAE34BD: public key "CentOS-7 Testing (CentOS 7 Testing content) <security@centos.org>" imported
Oct 16 10:22:35 vm9 esmith::event[6765]: gpg: key 352C64E5: public key "Fedora EPEL (7) <epel@fedoraproject.org>" imported
Oct 16 10:22:35 vm9 esmith::event[6765]: gpg: key 39BAF5C1: public key "NethServer 7 (NethServer 7 Official Signing Key) <security@nethserver.org>" imported
Oct 16 10:22:35 vm9 esmith::event[6765]: gpg: key 39BAF5C1: "NethServer 7 (NethServer 7 Official Signing Key) <security@nethserver.org>" not changed
Oct 16 10:22:35 vm9 esmith::event[6765]: gpg: Total number processed: 7
Oct 16 10:22:35 vm9 esmith::event[6765]: gpg:               imported: 6  (RSA: 6)
Oct 16 10:22:35 vm9 esmith::event[6765]: gpg:              unchanged: 1
Oct 16 10:22:35 vm9 esmith::event[6765]: gpg: inserting ownertrust of 6
Oct 16 10:22:35 vm9 esmith::event[6765]: gpg: inserting ownertrust of 6
Oct 16 10:22:35 vm9 esmith::event[6765]: gpg: inserting ownertrust of 6
Oct 16 10:22:35 vm9 esmith::event[6765]: gpg: inserting ownertrust of 6
Oct 16 10:22:35 vm9 esmith::event[6765]: gpg: inserting ownertrust of 6
Oct 16 10:22:35 vm9 esmith::event[6765]: gpg: inserting ownertrust of 6
Oct 16 10:22:35 vm9 esmith::event[6765]: Action: /etc/e-smith/events/software-repos-save/S21nethserver-base-import-repokeys SUCCESS [0.105251]
```

DavidePrincipi, 2019-10-16:
In nethserver-testing (all arches) nethserver-base-3.7.3-1.2.g15980bb.ns7.noarch.rpm

DavidePrincipi, 2019-10-16:
@markVnl, I'm sorry because I pointed you towards the wrong direction!

I found something that makes the applied patch useless.

It seems that repository metadata for CentOS Altarch are signed with the "official centos" key:
```text
gpg: Signature made Wed 02 Oct 2019 16:40:57 CEST using RSA key ID F4A80EB5
gpg: Good signature from "CentOS-7 Key (CentOS 7 Official Signing Key) <security@centos.org>"
```

For instance repomd.xml and repomd.xml.asc in http://mirror.centos.org/altarch/7/updates/armhfp/repodata/: the metadata signature is produced with an already known key...

So a step back to the original issue.

> `repomd.xml signature could not be verified for nethserver-base.`

My mistake, it's probably an oversight. This leads to a completely different problem.

Perhaps the import-repokeys didn't run, or run too early or too late?

DavidePrincipi, 2019-10-16:
@gsanchietti says that also a "clean-up" procedure for the ARM image could be the issue origin.

The GPG trustdb for repository metadata are stored under `/var/lib/yum`, (e.g. `/var/lib/yum/repos/x86_64/7/ce-base/gpgdir`. Do you remove it by some way?

markVnl, 2019-10-16:
> Perhaps the import-repokeys didn't run, or run too early or too late?

As mentioned i have difficulties debugging / testing it. Until now did not find the event which triggers this particular event.

> @gsanchietti says that also a "clean-up" procedure for the ARM image could be the issue origin.

I have tried to mitigate this for the RC1 by importing the nethserver-key on the rpi image @ image creation, this was after the issue popped up. And surly did not do so on the latest U-boot image.

Only cleanup up is done: `yum clean all` 
https://github.com/markVnl/nethserver-createimg/blob/master/create-image/ks/Nethserver-7.7.1908-RC1-RaspberryPi.ks#L280

hmm.. let me investigate some more..

markVnl, 2019-10-16:
> Perhaps the import-repokeys didn't run, or run too early or too late?

It probably did not run because nethserver-subscription is not installed.
https://github.com/NethServer/nethserver-subscription/blob/9a87d4e7e40ef07ea22877c9acfe9dc6b3ff055d/root/etc/e-smith/events/actions/nethserver-subscription-eorepo#L27

Although is is also called here:
 https://github.com/NethServer/nethserver-base/blob/19916a892eaf46ebf17c328ba44736011ef3ed6f/createlinks#L285

With the patch applies software center works well after:
`signal-event software-repos-save`

will try  jet again on a fresh image without the patch applied

DavidePrincipi, 2019-10-16:
There *must* be nethserver-subscription: it is part of the @nethserver-iso YUM group. Both the ISO and the `nethserver-install` command rely on that group composition to work.

So if the `import-repokeys` action is called by software-repos-save, called by nethserver-subscription-update, called by system-init ...we should not hit the bug :thinking:

markVnl, 2019-10-16:
> There _must_ be nethserver-subscription: it is part of the @nethserver-iso YUM group

The arm-iso group is a bit different, especially because we where never able to install nethserver-subscription until recently.. 

> will try jet again on a fresh image without the patch applied
No really seem to need the patch

DavidePrincipi, 2019-10-16:
I mounted the 2nd partition of the RB image (offline) and found a difference in the imported RPM keys. Only the GPG key `NethServer 7 (NethServer 7 Official Signing Key) <security@nethserver.org>` is present in that RPM db.

I'd expect to find more, because the x86_64 kickstart [try to import all the available keys](https://github.com/NethServer/nethserver-createiso/blob/5481c607e63061047f24be7916885a2513444104/src/lib/ks/unattended#L142).

markVnl, 2019-10-16:
This test image works:
https://drive.google.com/file/d/1H8xbooE9OA3Dq3jaWwEGv6WfEFZ1vmZW/view?usp=sharing

- installed nathserver-base with patch-applied from a local-repository
- added nethserver-subscription to the package list

> BTW, the `nethserver-install` command does the same: do we have the same issue by following the Install on CentOS procedure?
No we asume -y with yum install
https://github.com/NethServer/nethserver-release/blob/master/root/usr/sbin/nethserver-install#L129

DavidePrincipi, 2019-10-16:
As stated above, we do not need to import Arch SIG GPG key for YUM metadata, because their repositories metadata are signed with CentOS official signing key. Thus #162 cannot help.

I'm looking to this:

```text
[root@localhost ~]# rpm -q nethserver-subscription
package nethserver-subscription is not installed
[root@localhost ~]# rpm -q nethserver-base
nethserver-base-3.7.3-1.ns7.noarch
```

If I run `signal-event software-repos-save` repositories work correctly (no `yum -y` required)!

I guess the fix is in this diff and I bet it is the missing nethserver-subscription package in group nethserver-iso!

    diff -w -u <(curl http://mirror.nethserver.org/nethserver/7.7.1908/updates/x86_64/repodata/0f8c000659c9a8717ebcec705cb56c7611ee2bee3ad90edc0470693d9c03e89d-updates-groups.x86_64.xml) <(curl http://mirror.nethserver.org/nethserver/7.7.1908/updates/armhfp/repodata/7e6f257416f24cfa5833e06985b220a97c596e15ab09ad7b08775b75dcdc9b5e-updates-groups.armhfp.xml)

markVnl, 2019-10-16:
Proposals:

**1:**
- We apply the "extra key patch" to nethserver-base and can live with the _exit 2_ because of missing (extra) repo-key's. 
- Either install nethserver-subsriction by default (which happens any way if we decide to include nethserver-cockpit) or I hack an extra sim-link on @ image creation to include the software-repos-save event in systm-init

**2:**
- create a side package lets say _nethserver-arm-base_ carrying pretty much a copy of `/etc/e-smith/events/actions/nethserver-base-import-repokeys` merely containing the extra arm key's. (which "createlinks" in `software-repos-save` of course)
- Either install nethserver-subsriction by default (which happens any way if we decide to include nethserver-cockpit) or create a link in system-int with above package. (this could end up in running nethserver-base-import-repokeys twice @ system-init, IMHO small detail)

Not just because I'm lazy, but to avoid breakage in the furture favor (1)

markVnl, 2019-10-16:
> BTW I found `NetworkManager` running in the rp3: is it wanted? I'd [expect it disabled](https://github.com/NethServer/nethserver-createiso/blob/5481c607e63061047f24be7916885a2513444104/src/lib/ks/unattended#L147)

since 7.7 we need NetworkManger to kick start Network, it could be it is just because of `NetworkManager-wait-online.service` otherwise system-int starts before dhcp negotiation. 
On nethserver-7.6 we did not install NetworkManger. 
( it is one of those pieces of software I realy dislike - understatement - and do everything in my power or remove it. IIRC cockpit requires NM so gave up the resistance...)

DavidePrincipi, 2019-10-16:
Well NM has always been installed by Anaconda. The package is present in x86_64, however the service is set disabled.

DHCP (if configured by Anaconda) is started by `/etc/e-smith/events/system-init/S95nethserver-base-interface-update` during system-init, after all modules were configured.

markVnl, 2019-10-16:
> I think it's better to fix the comps!

Do not completely agree with this, an a arm system you want to be able to create a minimal system for one or two specic tasks . And I like to keep the this option available with nethserver-install

markVnl, 2019-10-16:
> DHCP (if configured by Anaconda) is started by `/etc/e-smith/events/system-init/S95nethserver-base-interface-update` during system-init, after all modules were configured.

explicitly added it otherwise i always end up with ip 192.168.1.1, maybe a ifcfg-eth0 will be sufficiant too.

DavidePrincipi, 2019-10-16:
> Do not completely agree with this

I understand your doubts, we have to keep it clean&simple: it is valid for every architecture!

I think we have also to limit the differences between architectures. 

1. I'd like to optimize disk space and features after a "stable" is released.
1. Cockpit should be installed. It is the future of Server Manager. We can improve it and its dependencies but this needs time and work.

> So if the import-repokeys action is called by software-repos-save, called by nethserver-subscription-update, called by system-init ...we should not hit the bug thinking

We probably we rely too much on nethserver-subscription. Maybe the correct way to find a compromise is to refactor nethserver-subscription and move some of its responsabilities to nethserver-base. This is not trivial. And I prefer an ARM image similar to x86_64.

> maybe a ifcfg-eth0 will be sufficiant too

system-init is a bad beast. The action

    /etc/e-smith/events/system-init/S50nethserver-base-initialize-db

extracts network settings written by Anaconda and stores them in e-smith NetworksDB.  We're probably missing them in the ARM image:

```text
[root@localhost ~]# grep 'green role not found' /var/log/messages 
Jan  1 00:00:34 localhost esmith::event[661]: [WARNING] green role not found. Using default IP configuration on eth0.
```

markVnl, 2019-10-17:
> Edit on my pi3 first boot I got an IP conflict with DHCP. Please note that the service is `network`, not `NetworkManager`

yes it used to work; 
[Nethserver-7.6.1810-Alpha-RaspberryPi.ks](https://github.com/markVnl/nethserver-createimg/blob/ccf13904bb1cd9a2858c2949539a0d81fc4f2def/create-image/ks/obsolete/Nethserver-7.6.1810-Alpha-RaspberryPi.ks#L11)
[Nethserver-7.7.1908-RC1-RaspberryPi.ks](https://github.com/markVnl/nethserver-createimg/blob/ccf13904bb1cd9a2858c2949539a0d81fc4f2def/create-image/ks/Nethserver-7.7.1908-RC1-RaspberryPi.ks#L11) 

_OT
I'm on the road to day so can't not reply so easily

DavidePrincipi, 2019-10-25:
@markVnl after applying the comps patch I forgot to check the repo metadata: the patch wasn't in the repositories! I just forced a metadata rebuild and now it has been propagated correctly:

http://packages.nethserver.org/nethserver/7.7.1908/updates/armhfp/repodata/

Could you prepare a new image from this change?

About NetworkManager, we can split that discussion to a separate issue.

markVnl, 2019-10-27:
> @markVnl after applying the comps patch I forgot to check the repo metadata: the patch wasn't in the repositories! I just forced a metadata rebuild and now it has been propagated correctly:

Confirmed, nethserver-subscription (among nethserver-cockpit) is installed throug the yum-comps @iso group

> About NetworkManager, we can split that discussion to a separate issue.

Yes will open an new issue

closed 2019-10-27 by markVnl