NethServer/arm-dev #34: Cockpit Dashboard empty hardware/info

dz00te, 2019-10-27:
Module: nethserver-cockpit

hardware/cpu fileds in Cockpit Dashboard  on ARM are empty. 
See also the previous issue #4  for NethGui

dz00te, 2019-10-27:
just adapted what done from @markVnl for nethgui, it seems to work...
https://github.com/dz00te/nethserver-cockpit/commit/a7cebcec83e404499a676c517d2661efbbfd54ef

gsanchietti, 2019-10-28:
Excellent, please open a PR! ;)

markVnl, 2019-10-28:
Yes this should work!
Now the question if it is accepted in "upstream" x86_64? 

/cc @gsanchietti @edospadoni 

PS: a little clean up off the code may be possible to:

The function readDMI() is called without an argument, meaning the _$id_ variable could be removed in the declaration of the function 
https://github.com/dz00te/nethserver-cockpit/blob/a7cebcec83e404499a676c517d2661efbbfd54ef/api/system-status/read#L74

dz00te, 2019-12-22:
fork synced and merged. still works on rpi/odroid/pine. before make a PR i would like to have some other tests on x86, i'll ask also on forum

DavidePrincipi, 2020-01-08:
@dz00te , any update?

dz00te, 2020-01-10:
i did not have feedbak from forum. i've just resynced my fork, and still seems ok to me... i would make the PR if you agree...

dz00te, 2020-01-15:
all ok with nethserver-cockpit-1.3.15-1. tnx

closed 2020-01-15 by dz00te