NethServer/arm-dev #35: Networkmanager should not run after system init

markVnl, 2019-10-27:
Since centos 7.7.1908 it is not trivial to reliable fetch a IP from a dhcp server with the network.service.

The network.service fails @boot. 
This appears to be caused  the service does not wait for the device (eth0) to be ready suggest you plug in a cable.. 

If you `systemctl restart network.service` after booting it brings up the network correctly.

the workaround in place is to start the Networkmanager service(s)

DavidePrincipi, 2019-10-27:
We could configure systemd to wait for the device to appear

https://bbs.archlinux.org/viewtopic.php?id=185352

markVnl, 2019-10-28:
> As alternative to NM, we could configure systemd to wait for the device to appear

Additional info:
```
# systemctl list-units | grep eth0
sys-devices-platform-scb-fd580000.genet-net-eth0.device  loaded active plugged   /sys/devices/platform/scb/fd580000.genet/net/eth0
sys-subsystem-net-devices-eth0.device                    loaded active plugged   /sys/subsystem/net/devices/eth0
```

DavidePrincipi, 2019-10-28:
Here's the command output from my rpi3

```text
[root@davidep2 ~]# systemctl list-units | grep eth0
sys-devices-platform-soc-3f980000.usb-usb1-1\x2d1-1\x2d1.1-1\x2d1.1:1.0-net-eth0.device                   loaded active plugged   SMSC9512/9514 Fast Ethernet Adapter
sys-subsystem-net-devices-eth0.device                                                                     loaded active plugged   SMSC9512/9514 Fast Ethernet Adapter
```

If we can assume on RPiX that the device name is always `eth0`, the unit `sys-subsystem-net-devices-eth0.device` can be set as dependency for `network.service`.

1. Create an override file for `network.service`: `systemctl edit network.service`
2. Add the following items:

       [Unit]
       After=sys-subsystem-net-devices-eth0.device
       Wants=sys-subsystem-net-devices-eth0.device

Here's the result

```text
Jan 01 00:00:09 davidep2.nethesis.it systemd[1]: Found device SMSC9512/9514 Fast Ethernet Adapter.
Jan 01 00:00:09 davidep2.nethesis.it systemd[1]: Starting LSB: Bring up/down networking...
```

See also the attached `systemd-analyze plot` output

[davidep2-networkboot.zip](https://github.com/NethServer/arm-dev/files/3777815/davidep2-networkboot.zip)

DavidePrincipi, 2019-10-28:
We have to consider also a use case where the Ethernet cable is unplugged and network is connected with a WiFi USB dongle or similar.

markVnl, 2019-10-28:
>     1. Create an override file for `network.service`: `systemctl edit network.service`

unfortunately it did not work.. 
I guess it could be even an driver issue on the rpi 4, although it puzzles me `NM` and` systemd-networkd ` brings up the network correctly n centos/nethserver 7 .

Have to attend other business will continue this evening

DavidePrincipi, 2019-10-28:
:cry: To ensure the overrides are in place, please check the output of `systemctl cat network.service`

This is my output
```text
[root@davidep2 ~]# systemctl cat network.service
# /run/systemd/generator.late/network.service
# Automatically generated by systemd-sysv-generator

[Unit]
Documentation=man:systemd-sysv-generator(8)
SourcePath=/etc/rc.d/init.d/network
Description=LSB: Bring up/down networking
Before=runlevel2.target
Before=runlevel3.target
Before=runlevel4.target
Before=runlevel5.target
Before=shutdown.target
Before=network-online.target
Before=network.target
After=iptables.service
After=ip6tables.service
After=NetworkManager-wait-online.service
After=NetworkManager.service
After=network-pre.target
Wants=network-online.target
Wants=network.target
Conflicts=shutdown.target

[Service]
Type=forking
Restart=no
TimeoutSec=5min
IgnoreSIGPIPE=no
KillMode=process
GuessMainPID=no
RemainAfterExit=yes
ExecStart=/etc/rc.d/init.d/network start
ExecStop=/etc/rc.d/init.d/network stop

# /etc/systemd/system/network.service.d/override.conf
[Service]
KillMode=control-group

[Unit]
After=sys-subsystem-net-devices-eth0.device
Wants=sys-subsystem-net-devices-eth0.device
```

markVnl, 2020-05-06:
for future reference workarounds in place on 7.8.2003

- /etc/sysconfig/network-scripts/ifcfg-eth0 configuration to kickstart networking on eth0
- sytemd-unit overide for network.service to require sys-subsystem-net-devices-eth0.device
- sytemd-unit overide for nethserver-system-init.service to require sys-subsystem-net-devices-eth0.device

Closing now with the promise to revised at the next point release ;)

markVnl, 2020-12-23:
> Closing now with the promise to revised at the next point release ;)

Although this issue was revisited before the release of the RPI 7.9.2009 RC1 image, could not think of a better solution..

It is getting grufty .. also see #46 

**Dropping a new idea here:** 

Enable Networkmanager and Networkmanager-wait-online for the first boot. After all NM does a better job in finding a working network connection than all the mitigations we though of.

Disable NM (and other well-known unwanted services during systen-init, like we do @[nethserver-install](https://github.com/NethServer/nethserver-release/blob/55132ed6eccdac15b3b53632be8daad21cfcbcdc/root/usr/sbin/nethserver-install#L133)

Something like this:
https://github.com/markVnl/nethserver-arm-extra-config/commit/9f6b73f4e13ba2235baa20122f4b00ef5ea7622c

markVnl, 2020-12-24:
After several new installs on several RPI's always ended up with:

    a functional network configuration
    a valid expire date for the server-certificate
    no complains from rrdtool/collectd in syslogs

closed 2020-12-24 by markVnl