NethServer/arm-dev #47: Update to nethsever-mail version 2.19.3 fails

markVnl, 2020-12-17:
Cause:
 nethserver-httpd-admin-service missing in base repo armhfp and aarch64 :

```
# yum update
...
...
--> Finished Dependency Resolution
Error: Package: nethserver-mail-filter-2.19.3-1.ns7.noarch (nethserver-updates)
           Requires: nethserver-httpd-admin-service
 You could try using --skip-broken to work around the problem
 You could try running: rpm -Va --nofiles --nodigest
```

The arm-base package repositories are not in sync, <br> arm is still on `nethserver-httpd-admin-2.5.1` resulting in missing  `nethserver-httpd-admin-service` .

x86_64 repo:
```
nethserver-httpd-admin-2.7.0-1.ns7.noarch.rpm
nethserver-httpd-admin-service-2.7.0-1.ns7.noarch.rpm
```

cc @DavidePrincipi @gsanchietti

markVnl, 2020-12-20:
Manually uploaded to armhfp-testing:
```
nethserver-httpd-admin-2.7.0-1.ns7.noarch.rpm
nethserver-httpd-admin-service-2.7.0-1.ns7.noarch.rpm
```

No errors after: ` yum --enablerepo=nethserver-testing update nethserver-httpd-admin*`

After this update of **nethserver-mail-\*-2.19.3-1.ns7.noarch** succeeded.

```
# rpm -qa |grep nethserver-mail
nethserver-mail-server-2.19.3-1.ns7.noarch
nethserver-mail-smarthost-2.19.3-1.ns7.noarch
nethserver-mail-filter-2.19.3-1.ns7.noarch
nethserver-mail-common-2.19.3-1.ns7.noarch
```

DavidePrincipi, 2020-12-21:
I run a diff against ARM and x86_64 "base" repo and found a bunch of RPMs were missing. I pushed the following list of noarch RPMs to aarch64 and armhfp "updates" repo:

freepbx-14.0-11.ns7.noarch.rpm
nethserver-backup-data-1.7.4-1.ns7.noarch.rpm
nethserver-cgp-2.4.0-1.ns7.noarch.rpm
nethserver-cockpit-1.7.12-1.ns7.noarch.rpm
nethserver-cockpit-lib-1.7.12-1.ns7.noarch.rpm
nethserver-evebox-1.4.0-1.ns7.noarch.rpm
nethserver-fail2ban-1.5.6-1.ns7.noarch.rpm
nethserver-firewall-base-3.13.0-1.ns7.noarch.rpm
nethserver-firewall-base-ui-3.13.0-1.ns7.noarch.rpm
nethserver-httpd-3.11.4-1.ns7.noarch.rpm
nethserver-httpd-admin-2.7.0-1.ns7.noarch.rpm
nethserver-httpd-admin-service-2.7.0-1.ns7.noarch.rpm
nethserver-httpd-proxypass-3.11.4-1.ns7.noarch.rpm
nethserver-httpd-virtualhosts-3.11.4-1.ns7.noarch.rpm
nethserver-lang-cockpit-1.4.6-3.ns7.noarch.rpm
nethserver-lang-cs-1.4.6-3.ns7.noarch.rpm
nethserver-lang-de-1.4.6-3.ns7.noarch.rpm
nethserver-lang-el-1.4.6-3.ns7.noarch.rpm
nethserver-lang-en-1.4.6-3.ns7.noarch.rpm
nethserver-lang-es-1.4.6-3.ns7.noarch.rpm
nethserver-lang-fr-1.4.6-3.ns7.noarch.rpm
nethserver-lang-hu-1.4.6-3.ns7.noarch.rpm
nethserver-lang-it-1.4.6-3.ns7.noarch.rpm
nethserver-lang-nl-1.4.6-3.ns7.noarch.rpm
nethserver-lang-pt-1.4.6-3.ns7.noarch.rpm
nethserver-lang-ro-1.4.6-3.ns7.noarch.rpm
nethserver-lang-ru-1.4.6-3.ns7.noarch.rpm
nethserver-lang-sr-1.4.6-3.ns7.noarch.rpm
nethserver-lang-sv-1.4.6-3.ns7.noarch.rpm
nethserver-lang-tr-1.4.6-3.ns7.noarch.rpm
nethserver-lightsquid-1.2.0-1.ns7.noarch.rpm
nethserver-mail-common-2.19.2-1.ns7.noarch.rpm
nethserver-mail-disclaimer-2.19.2-1.ns7.noarch.rpm
nethserver-mail-filter-2.19.2-1.ns7.noarch.rpm
nethserver-mail-getmail-2.19.2-1.ns7.noarch.rpm
nethserver-mail-imapsync-2.19.2-1.ns7.noarch.rpm
nethserver-mail-ipaccess-2.19.2-1.ns7.noarch.rpm
nethserver-mail-p3scan-2.19.2-1.ns7.noarch.rpm
nethserver-mail-quarantine-2.19.2-1.ns7.noarch.rpm
nethserver-mail-server-2.19.2-1.ns7.noarch.rpm
nethserver-mail-smarthost-2.19.2-1.ns7.noarch.rpm
nethserver-netdata-2.0.0-1.ns7.noarch.rpm
nethserver-nextcloud-1.15.0-1.ns7.noarch.rpm
nethserver-ntopng-3.1.0-1.ns7.noarch.rpm
nethserver-openssh-1.8.0-1.ns7.noarch.rpm
nethserver-php-1.3.0-1.ns7.noarch.rpm
nethserver-squid-1.10.14-1.ns7.noarch.rpm
nethserver-subscription-3.6.6-1.ns7.noarch.rpm
nethserver-subscription-ui-3.6.6-1.ns7.noarch.rpm
nethserver-suricata-2.3.1-1.ns7.noarch.rpm
nethserver-vpn-ui-1.5.5-1.ns7.noarch.rpm

markVnl, 2020-12-21:
Thank you, pretty sure this resolves this issue !

As they say don't assume: test, test and test ;) , will do so today/this evening and if nothing come's up going to close this issue,

markVnl, 2020-12-21:
Sorry to Bother you one more :( @DavidePrincipi 

The (arm-specific) groups (COMPS) seem to be missing now (on armhfp):

```
# yum groups list
Loaded plugins: changelog, fastestmirror, nethserver_events
Loading mirror speeds from cached hostfile
 * ce-base: mirrors.coreix.net
 * ce-extras: mirrors.coreix.net
 * ce-updates: mirrors.coreix.net
 * centos-kernel: mirror.nl.datapacket.com
Installed Environment Groups:
   Minimal Install
Available Environment Groups:
   Compute Node
   Infrastructure Server
   File and Print Server
   Basic Web Server
   Virtualization Host
   Server with GUI
   GNOME Desktop
   KDE Plasma Workspaces
   Development and Creative Workstation
Installed Groups:
   Legacy UNIX Compatibility
   Security Tools
   System Administration Tools
Available Groups:
   Compatibility Libraries
   Console Internet Tools
   Development Tools
   Graphical Administration Tools
   Scientific Support
   Smart Card Support
   System Management
Done

Note it does not show "nethserver-groups";
The software-center is also empty..  
```

markVnl, 2020-12-21:
(Before you ask) Groups are fixed on armhfp

```
#  yum --disablerepo=* --enablerepo=nethserver-updates groups list
Loaded plugins: changelog, fastestmirror, nethserver_events
Loading mirror speeds from cached hostfile
Installed Groups:
   Email
   New Server Manager (Beta)
   Statistics
Available Groups:
   Backup
   Basic firewall
   Dedalo Hotspot
   Dutch language
   FTP server
   Fail2ban
   File server
   French language
   German language
   Greek language
   Hungarian language
   IPsec tunnels
   Intrusion Prevention System
   Italian language
   MariaDB (MySQL) server
   OpenVPN
   POP3 connector
   POP3 proxy
   Portuguese language
   Reverse proxy
   Romanian language
   Roundcube web mail
   Russian language
   SMTP proxy
   SNMP server
   Serbian language
   Spanish language
   Turkish language
   Web filter
   Web proxy
   Web server
Done

```

DavidePrincipi, 2020-12-21:
Please clean the caches and try again now!

markVnl, 2020-12-21:
Groups are oke; also software-center is populated again:

```
# yum clean all
Loaded plugins: changelog, fastestmirror, nethserver_events
Cleaning repos: ce-base ce-extras ce-sclo-rh ce-sclo-sclo ce-updates
              : centos-kernel epel nethserver-base nethserver-updates

# yum makecache
Loaded plugins: changelog, fastestmirror, nethserver_events

<Truncated>

Metadata Cache Created

# yum --disablerepo=* --enablerepo=nethserver-updates groups list
Loaded plugins: changelog, fastestmirror, nethserver_events
Loading mirror speeds from cached hostfile
Installed Groups:
   Email
   New Server Manager (Beta)
   Statistics
Available Groups:
   Backup
   Basic firewall
   Dedalo Hotspot
   Dutch language
   FTP server
   Fail2ban
   File server
   French language
   German language
   Greek language
   Hungarian language
   IPsec tunnels
   Intrusion Prevention System
   Italian language
   MariaDB (MySQL) server
   OpenVPN
   POP3 connector
   POP3 proxy
   Portuguese language
   Reverse proxy
   Romanian language
   Roundcube web mail
   Russian language
   SMTP proxy
   SNMP server
   Serbian language
   Spanish language
   Turkish language
   Web filter
   Web proxy
   Web server
Done
```

markVnl, 2020-12-21:
Still need to test aarch64, `respond.xml` in repo looks good!

Closing now, and will reopen if something comes-up with aarh64,

closed 2020-12-21 by markVnl