NethServer/arm-dev #48: Groups are missing after uploads to updates and nethforge repo's

markVnl, 2020-12-28:
After uploading nethserver-arm-extra-config-1.1.0-1 to armhfp and aarch64 the groups in the updates repo’s are missing again. The groups are also missing in nethforge armhfp and aarch64 repo’s after uploading nethserver-sogo and it’s dependencies to those repositories…

markVnl, 2021-01-11:
@DavidePrincipi not in a particular hurry :) 

Is there a status update on this?

DavidePrincipi, 2021-01-11:
I should look at the repo scripts in the next days, to analyze this issue and also evaluate GitHub actions as alternative to TravisCI.com

DavidePrincipi, 2021-01-13:
Hi Mark I think I found the issue origin. Please try again the upload... :crossed_fingers:

markVnl, 2021-01-13:
Yes can confirm it worked:

After uploading nethserver-arm-extra-config-1.1.2-1 to armhfp and aarch64 groups files are present!

Thanks

closed 2021-01-13 by markVnl